Repl1catorCZ/Predator-Delta-3D-Printer #13: Adaptor plate

totalitarian, 2019-05-23:
Do we need to print an adaptor plate to mount the board on the existing standoffs?

Repl1catorCZ, 2019-05-24:
Hi, Thanks for report. 
Yes you need print a case for board. I doesent have a fit one yet .. just an old one which i use for MKS gen L i will add stl ASAP ..

totalitarian, 2019-05-24:
Thanks. Just found one in the facebook group that i'm trailing now.

Repl1catorCZ, 2019-05-24:
@totalitarian if you find or model something awesome we could implement it on base stl .. im not at home this week then i cant help .. but most of  MKS gen L v1 are fit on skr .. but  try to post here some links for thingi .. but better way will be if we model one for us .. after a get home i mesure board and offset for perfect fit ..

totalitarian, 2019-05-24:
Just modelling one now. I'll report back if it's any good

closed 2019-06-04 by Repl1catorCZ